egguy/addon-grott #9: sensor.xxxxxxxxx_generated_energy_today caused a huge negative reading

MaoZ1993, 2023-10-12:
Hi Egguy

Thanks for your great work! This addon just made grott much much easier to deploy.

I added the sensor.xxxxxxxxx_generated_energy_today to the HA Energy panel and found it's been recognised a huge negative reading on every morning. It seems when inverter powered up and then the senser is resetting to zero, which maybe the cause of this wrong reading.

Is that possible to reset this sensor on everyday 0:00 to avoid the energy dashboard reading issue? Thanks.

[image]

MaoZ1993, 2023-10-12:
Oh Sorry

I suddenly realised I need to use Total instead of Today in the energy panel.... Sorry, please close this ticket...

egguy, 2023-10-13:
No worries. I need to search if there is a way to prevent HA to suggesting the daily total instead of the total.

closed 2023-10-13 by egguy

Tabularasa84, 2023-10-15:
I had also taken the Today Total entity. With me the negative value was always the total value of the day before. Then I will try the total entity right away. 
Edit: unfortunately this does not work with my Energy dashboard. Because I previously used the official Growatt integration and there also used the Today entity. Now I don't have any stats there unfortunately. Is there a way to solve this?

But thanks for the addon, otherwise I would have no more values.