egguy/addon-grott #27: Add-on fails to start

mrfixit1, 2024-02-12:
All fine until yesterday, today the add-on doesn’t start.

Info and error message from log:
…
 Add-on version: 0.1.7
 You are running the latest version of this add-on.
 System: Home Assistant OS 11.5  (aarch64 / raspberrypi3-64)
 Home Assistant Core: 2024.2.1
 Home Assistant Supervisor: 2024.01.1
….
s6-rc: info: service legacy-services successfully started
[13:43:24] INFO: Preparing to start...
[13:43:48] ERROR: Got unexpected response from the API: Service not enabled
./script.sh: line 230: GROTT_CONFIG_MQTT_HOST: unbound variable
s6-rc: info: service legacy-services: stopping
…

The datalogger shows CONN OK. Switching it from HA back to the direct connection with server.growatt.com updates the data in the Growatt dashboard, as expected.

Anyone have any ideas how to get the add-on going again?

egguy, 2024-02-12:
Hi, could you try to restart the mosquitto addon, then the add-on?

It seems this procedure solve this error. I still don't know the root cause, I will try to find a more definitive fix.

mrfixit1, 2024-02-12:
Thanks for the super quick response. Unfortunately the restarts didn’t help. The error message is no longer present in the grott add-on log and all looks normal up to the first response from the Growatt server, but nothing more appears and I’m not seeing any PV data in the HA dashboard.

mrfixit1, 2024-02-13:
Daytime update. No data this morning after the 2nd restart yesterday, or after a 3rd cycle of: stop grott - restart mosquitto - start grott. I then set the data logger back to the growatt server, stopped grott, restarted mosquitto, started grott and reset the data logger to HA; this worked!

closed 2024-03-22 by github-actions[bot]

splitbrain, 2024-04-10:
I had the same issue. A restart of mosquitto _did_ fix it for me.