egguy/addon-grott #32: Add-on stops working infrequently at random times

mrfixit1, 2024-03-05:
The add-on stopped unexpectedly again yesterday. When I checked my PV performance after dark, the output power was stuck at the value from around 15:00 and the energy dashboard showed less solar production (from Grott) than sent to the grid (from the power meter). Data gaps have happened before, maybe about twice a month on average. I previously thought they were only after HA updates, but this one however was not. This morning the usual workaround of restarting the add-on restored the data flow.

Config: Grott add-on v0.1.7, HAOS core 2024.2.4 on a Pi3.

lexo12345, 2024-03-30:
I'm having the same issue. It was all working & then got stuck/stopped updating data/sensors.

Any ideas on how to diagnose/fix?

Gertjezzz, 2024-04-12:
I'm having the same issue. Manually restart addon helps starting en reconnecting data. It happens multiple times a day, where my tracking of data is lost.

Dont know if this helps, but its the last lines of the log before exiting service;`  

> proxy.main(conf)
>   File "/app/grottproxy.py", line 144, in main
>     self.on_recv(conf)
>   File "/app/grottproxy.py", line 238, in on_recv
>     procdata(conf,data)    
>     ^^^^^^^^^^^^^^^^^^^
>   File "/app/grottdata.py", line 220, in procdata
>     for keyword in  conf.recorddict[layout].keys() :
>                     ~~~~~~~~~~~~~~~^^^^^^^^
> KeyError: 'T06NNNNMIN'
> s6-rc: info: service legacy-services: stopping
> s6-rc: info: service legacy-services successfully stopped
> s6-rc: info: service legacy-cont-init: stopping
> s6-rc: info: service legacy-cont-init successfully stopped
> s6-rc: info: service fix-attrs: stopping
> s6-rc: info: service base-addon-log-level: stopping
> s6-rc: info: service fix-attrs successfully stopped
> s6-rc: info: service base-addon-log-level successfully stopped
> s6-rc: info: service base-addon-banner: stopping
> s6-rc: info: service base-addon-banner successfully stopped
> s6-rc: info: service s6rc-oneshot-runner: stopping
> s6-rc: info: service s6rc-oneshot-runner successfully stopped`

egguy, 2024-04-15:
Ho , thanks for the debug log. 

I'm going to take a look at it.

marcelwestrahome, 2024-05-02:
Same for me, turn on the watch dog, but does not make sense

Gertjezzz, 2024-06-02:
The issue hasn't been resolved. Updated all to the most recent version, even beta version. Didn't fix the issue

egguy, 2024-06-04:
Hi, there is a new update as the new stable branch is the 2.8.3, it should be better.

If it continues, Would you mind giving me more logs? This would help me to discover if it is a grott bug who should be reported upstream/

Gertjezzz, 2024-06-04:
Thanks for the reply. I did an update. But I could not find a stable 2.8.3 in the addon store.
I have 2 Beta 2.8.x versions and a stable 2.7 version.
[image]

Here is the log of the 2.8.x beta version;
`
	 - Growatt packet received:
		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.9', 44438), raddr=('47.91.67.66', 5279)>
	 - Growatt data decrypted V2
	 - Matched inverter serial to inverter type MIN
Traceback (most recent call last):
  File "/app/grott.py", line 35, in <module>
    proxy.main(conf)
  File "/app/grottproxy.py", line 144, in main
    self.on_recv(conf)
  File "/app/grottproxy.py", line 238, in on_recv
    procdata(conf,data)    
    ^^^^^^^^^^^^^^^^^^^
  File "/app/grottdata.py", line 220, in procdata
    for keyword in  conf.recorddict[layout].keys() :
                    ~~~~~~~~~~~~~~~^^^^^^^^
KeyError: 'T06NNNNMIN'
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting

-----------------------------------------------------------
 Add-on: Grott Beta branch (2.8)
 The Growatt inverter monitor with current HA plugin integrated
-----------------------------------------------------------
 Add-on version: 0.1.7
 You are running the latest version of this add-on.
 System: Home Assistant OS 12.3  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2024.5.5
 Home Assistant Supervisor: 2024.05.1
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[17:44:37] INFO: Preparing to start... '

Gertjezzz, 2024-06-04:
And the log of the 2.7 stable version

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.9', 60320), raddr=('47.91.67.66', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.9', 60320), raddr=('47.91.67.66', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.9', 60308), raddr=('47.91.67.66', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.9', 60320), raddr=('47.91.67.66', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.9', 60320), raddr=('47.91.67.66', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.9', 60308), raddr=('47.91.67.66', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.9', 60320), raddr=('47.91.67.66', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.9', 60320), raddr=('47.91.67.66', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.9', 60308), raddr=('47.91.67.66', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.9', 60320), raddr=('47.91.67.66', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.9', 60320), raddr=('47.91.67.66', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=7, family=2, type=1, proto=0, laddr=('172.30.33.9', 5279), raddr=('192.168.178.14', 1026)>

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.9', 60320), raddr=('47.91.67.66', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.9', 60308), raddr=('47.91.67.66', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.9', 60320), raddr=('47.91.67.66', 5279)>
	 - Growatt data decrypted V2

	 - Growatt packet received:
		  <socket.socket fd=7, family=2, type=1, proto=0, laddr=('172.30.33.9', 5279), raddr=('192.168.178.14', 1026)>

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.9', 60320), raddr=('47.91.67.66', 5279)>
	 - Growatt data decrypted V2
	 - Matched inverter serial to inverter type TL3
	 - Grott MQTT topic used : energy/growatt

	 - Growatt packet received:
		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.9', 60308), raddr=('47.91.67.66', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=7, family=2, type=1, proto=0, laddr=('172.30.33.9', 5279), raddr=('192.168.178.14', 1026)>

	 - Growatt packet received:
		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.9', 60308), raddr=('47.91.67.66', 5279)>
	 - Growatt data decrypted V2
	 - Matched inverter serial to inverter type MIN
Traceback (most recent call last):
  File "/app/grott.py", line 35, in <module>
    proxy.main(conf)
  File "/app/grottproxy.py", line 144, in main
    self.on_recv(conf)
  File "/app/grottproxy.py", line 238, in on_recv
    procdata(conf,data)    
    ^^^^^^^^^^^^^^^^^^^
  File "/app/grottdata.py", line 220, in procdata
    for keyword in  conf.recorddict[layout].keys() :
                    ~~~~~~~~~~~~~~~^^^^^^^^
KeyError: 'T06NNNNMIN'
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting

-----------------------------------------------------------
 Add-on: Grott stable branch (2.7)
 The Growatt inverter monitor with current HA plugin integrated
-----------------------------------------------------------
 Add-on version: 0.1.9
 You are running the latest version of this add-on.
 System: Home Assistant OS 12.3  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2024.5.5
 Home Assistant Supervisor: 2024.05.1
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting

egguy, 2024-07-22:
Bug has been identified, I created a bug in upstream.

Thanks for your patience

lexo12345, 2024-07-25:
Can someone please help me fix this issue?

I have been trying for months to get this back up & running & I only have limited time when I get home as the inverter turns off when its dark.

I have set:
ggrowattip 8.209.71.240
gnoipf on
gblockcmd on

pointed the datalogger back at my HA ip address

Unsure if I'm even getting data into grott, log isn't very helpful.

Please let me know what information you need to see to try guide me through this. This was working before this bug, but not I've tried changing so much that I'm lost.

mrfixit1, 2024-07-26:
@lexo12345 is 8.209.71.240 the correct growatt ip for your location? Check it from a local terminal, e.g. on a Mac with: dscacheutil -q host -a name server.growatt.com
What do you mean by "log isn’t very helpful”? Does it show “packet received” messages similar to the start of (https://github.com/egguy/addon-grott/issues/32#issuecomment-2147902592) above?

lexo12345, 2024-07-27:
> @lexo12345 is 8.209.71.240 the correct growatt ip for your location? Check it from a local terminal, e.g. on a Mac with: dscacheutil -q host -a name server.growatt.com What do you mean by "log isn’t very helpful”? Does it show “packet received” messages similar to the start of ([#32 (comment)](https://github.com/egguy/addon-grott/issues/32#issuecomment-2147902592)) above?

Thankyou for your help. Got it back up and running!

Did a reverse DNS & found I was using the wrong IP.

Also put a space instead of a dot on the datalogger IP which didn't help 😂

fernandoacf, 2024-07-28:
I'm trying to set these options on the configurations page, but it looks like the configurations aren't set. still getting all default values. is there any catch do change configurations?

mrfixit1, 2024-10-20:
The last few weeks the add-on has stopped working regularly, one or more times per week. A manual restart always brings it back to life, but it would be better if this wasn’t necessary. I wonder if an automation or cron job might be possible to restart it every night.

rdpeake, 2024-12-13:
I keep running into issues where my datalogger stops communicating with grott - sometimes a restart of both the add on and the datalogger will fix the issue - but often it won't. At one point i had to disable DHCP on the shinelink box as it was flooding my network with DHCP requests. do i have a fault with the shinelink box, or is there something within grott that is causing an issue on my shinelink?

malteschlueter, 2025-02-27:
Does anyone have an update on this issue?
I also have the problem that the add-on or Grott does not process any data. As soon as I restart the add-on, the data is sent again as normal.

I am currently considering creating an automation as a workaround that automatically restarts the add-on if the data is older than 5-10 minutes

rdpeake, 2025-02-27:
I'm currently using the time since commisioned variable not changing for 10 minutes I think I picked as a template problem entity, and then running an automation every 30 until that variable is resolved. I swear the issue is on Trott server side - but when the issue is happening accessing the admin portal of the shine link box is very hard.