egguy/addon-grott #56: PV data no longer arriving in Home Assistant

mrfixit1, 2024-07-16:
Sometime today PV data stopped arriving in HA. Somehow my Growatt datalogger had lost its rerouting settings. Restoring them (i.e. ResolvDomain to OFF and Server IP = HA LAN IP) only worked for a few minutes, after which they regressed back to the defaults. I eventually fixed this with a hard reboot of the datalogger and another restore, after which the settings appear to be persistent again. 

However I’m still not seeing any PV data in HA. I tried restarting the grott add-on, the sequence in https://github.com/egguy/addon-grott/issues/27#issuecomment-1940781080, reloading HA, rebooting HA, reinstalling grott and trying the beta version, all without success. After every restart of the grott add-on, the log ends at:
Hostname : 8ce4abae-grott
IP :  172.30.33.3 , port :  5279

Very similar to #32 and #27, but so far without a workaround. Ideas welcome!

Config: Grott stable branch (2.8) v0.1.10, HA core 2024.7.2, Supervisor 2024.06.2, HAOS 12.4 on a Pi3.

sroka222, 2024-07-16:
I have the same problem. Also from today.

egguy, 2024-07-16:
Hi, It seems to come from growatt forcing a reset of the sticks to the new server IP. 

A temporary workaround is to update the growatt server IP in the configuration. You can find the procedure here: https://github.com/egguy/addon-grott-beta/issues/51#issuecomment-2231910621

mrfixit1, 2024-07-17:
Thanks for the quick reply and tip Etienne, the workaround worked for me. I’m closing this issue as the fix will be handled by https://github.com/johanmeijer/grott/pull/565.

closed 2024-07-17 by mrfixit1