egguy/addon-grott #58: Grott does not talk with mqtt broker

Gemies, 2024-07-17:
Since yesterday morning 8:30 grott is not talking to mqtt anymore. Tried several restarts and reconfiguratio n of grott mqtt broker and mqtt integration within home assistant.

Can you please take al look at my logfiles?

	 - Growatt packet received:
		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.6', 55090), raddr=('47.91.67.66', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.6', 55090), raddr=('47.91.67.66', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.6', 55090), raddr=('47.91.67.66', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.6', 55090), raddr=('47.91.67.66', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.6', 55090), raddr=('47.91.67.66', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.6', 55090), raddr=('47.91.67.66', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.6', 55090), raddr=('47.91.67.66', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.6', 55090), raddr=('47.91.67.66', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.6', 55090), raddr=('47.91.67.66', 5279)>
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting
-----------------------------------------------------------

 Add-on: Grott stable branch (2.8)
 The Growatt inverter monitor with current HA plugin integrated
-----------------------------------------------------------
 Add-on version: 0.1.10
 You are running the latest version of this add-on.
 System: Home Assistant OS 12.4  (amd64 / qemux86-64)
 Home Assistant Core: 2024.7.2
 Home Assistant Supervisor: 2024.06.2
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[13:57:22] INFO: Preparing to start...
	 **********************************************************************************
	 - Grott - libscrc not installed, no CRC checking only record validation on length!
	 **********************************************************************************
Grott Growatt logging monitor : 2.8.3

Grott process configuration file

Grott process environmental variables

	Pulled 'ginvtype=default' from the environment

	Pulled 'ginvtypemap="NAC5A420T5"' from the environment

	Pulled 'gnomqtt=False' from the environment

	Pulled 'gmqttip=172.30.33.0' from the environment

	Pulled 'gmqttport=1883' from the environment

	Pulled 'gmqttauth=True' from the environment

	Pulled 'gmqttuser=addons' from the environment

	Pulled 'gmqttpassword=AeshieCeiWae1iDohJ0eeve0ohch0zais9niebeewe9JakuXaivaedu8eip7rai8' from the environment

Grott override settings if set in commandline

Grott process json layout files

Grott proxy mode started
	 **********************************************************************************
	 - Grott - libscrc not installed, no CRC checking only record validation on length!
	 **********************************************************************************
Hostname : 8ce4abae-grott
IP :  172.30.33.6 , port :  5279

egguy, 2024-07-17:
Hi, if the Growatt portal reports data, but you see nothing from the addon it could be related to: https://github.com/egguy/addon-grott-beta/issues/51#issuecomment-2231910621

Gemies, 2024-07-17:
> Hi, if the Growatt portal reports data, but you see nothing from the addon it could be related to: https://github.com/egguy/addon-grott-beta/issues/51#issuecomment-2231910621

After adding the ggrowattip and port, as suggested , i get this error in the logging of grott.

Hostname : 8ce4abae-grott
IP :  172.30.33.6 , port :  5279 

	 - Grott - grottproxy forward error :  [Errno 111] Connection refused
	 - Grott - grottproxy forward error :  [Errno 111] Connection refused
	 - Grott - grottproxy forward error :  [Errno 111] Connection refused
s

thoompje, 2024-08-07:
> After adding the ggrowattip and port, as suggested , i get this error in the logging of grott.
>
> Hostname : 8ce4abae-grott
> IP :  172.30.33.6 , port :  5279
>
> 	 - Grott - grottproxy forward error : [...]

I suddenly got this error also, nothing changed at my envoirment...

closed 2024-09-14 by github-actions[bot]