egguy/addon-grott #74: Wrong power reading HomeAssistant

sirox01, 2024-07-30:
After the update the power shown on Home Assistant reach above 400.000.000W every day when inverter shuts down.
This reading is not visible on the Growatt website and not on PVOutput.

Can anybody help?

[image]

sirox01, 2024-08-30:
I still have this problem every day.

typxxi, 2024-08-31:
time to show more details like the production of a whole day.

I do not see any value and it would be nice to use the tools provided. 
The sensor says it was still in production, but the graph did not show much. 
So use the time span from 8 to 6 to show the data if those are clean.

At the end if there is an issue then it is caused from the inverter way beyond daylight time when the inverter usually is shut down. 

There is a message arriving in grott and you should provide a bit more than just that.
Mos likely it is for sure not a grott ha addon issue but a grott core issue or it is cause on your side and getting into the system via mqtt. 
You can find these messages in mqtt explorer if you look for that or wait for it to arrive. 
And then it is in the protocol / log.

But if you simply provide nothing do not expect that anyone will help you again.
A log is a basic and the documentation is well written and I mean the grott documentation provided here and the grott solutin by Johan Meier. 
It is your job to read the help, manual and what not first and if you would do so then you would read quite often what has to be checked or provided.

Now I have lost the 10 minutes just to let you know about the abc. 

And if you wait another month to ask again no one will care cause you were not able to provide more details after 1 month ?

Then you can not expect others to help you out. Sorry, but that is what a mirror would tell you too.

typxxi, 2024-08-31:
and of cause: 

I do not have such issue since the update.
Maybe luck but do you know anyone else who has 8 Growatt inverters from 3 different model lines running on a single HA grott installation (Pi 4)

Therefore I doubt that I had just luck. I do not think that it has been caused by egguys grott addon and if so you have to ask Johann meier cause the base is grott developed by Johan. 
Eggyuy is only porting it into the HA world afaik.

BTW: you  did not even mention your type of inverter and all the details that might be usefull. 
Might be worth to mention that you could have easily setup a HA test installation with grott to cross check if that happens in the testbed too with the previous version of grott addon and then with the most recent one.

You had 1 month time and you have provided nothing to help your issue.

Why should someone do your job like I did now ?
And even if you post it on Meiers github project no one would support such superficial written issue.

sirox01, 2024-08-31:
I have made this question with Johan Meijer.
He is an Dutch person who is active on a tech site called Tweakers.
I gave him my problem and he told me to make a topic on the github from the addon.

I'm sorry I didn't provide with more information but I was hoping I wasn't the only one with the problem.

De logs from addon:

Grott stable branch (2.8)
- Growatt packet received:
<socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.4', 42768), raddr=('8.209.71.240', 5279)>
- Growatt data decrypted V2
- Grott MQTT topic used : energy/growatt

- Growatt packet received:
<socket.socket fd=7, family=2, type=1, proto=0, laddr=('172.30.33.4', 5279), raddr=('192.xxx.xxx.xx', 1029)>

- Growatt packet received:
<socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.4', 42768), raddr=('8.209.71.240', 5279)>

Logs MQTT:

Bericht 161 ontvangen op homeassistant/grott/TDG/state om 21:23:
{"pvserial": "TDG", "pvstatus": 1, "pvpowerin": 1, "pv1voltage": 867, "pv1current": 0, "pv1watt": 1, "pv2voltage": 1142, "pv2current": 0, "pv2watt": 0, "pv3voltage": 0, "pv3current": 0, "pv3watt": 0, "pv4voltage": 0, "pv4current": 0, "pv4watt": 0, "pvpowerout": 4294967295, "pvfrequentie": 5001, "pvgridvoltage": 2342, "pvgridcurrent": 6, "pvgridpower": 151, "pvgridvoltage2": 0, "pvgridcurrent2": 0, "pvgridpower2": 0, "pvgridvoltage3": 0, "pvgridcurrent3": 0, "pvgridpower3": 0, "vacrs": 2342, "vacst": 0, "vactr": 0, "ptousertotal": 0, "ptogridtotal": 0, "ptoloadtotal": 0, "totworktime": 99151306, "pvenergytoday": 179, "pvenergytotal": 99671, "epvtotal": 103387, "epv1today": 100, "epv1total": 55915, "epv2today": 86, "epv2total": 47472, "epv3today": 0, "epv3total": 0, "etousertoday": 0, "etousertotal": 0, "etogridtoday": 0, "etogridtotal": 0, "eloadtoday": 0, "eloadtotal": 0, "deratingmode": 0, "iso": 65530, "dcir": 0, "dcis": 0, "dcit": 0, "gfci": 0, "pvtemperature": 399, "pvipmtemperature": 399, "temp3": 0, "temp4": 0, "temp5": 486, "pbusvoltage": 3592, "nbusvoltage": 0, "ipf": 20000, "realoppercent": 0, "opfullwatt": 0, "standbyflag": 0, "faultcode": 0, "warningcode": 0, "systemfaultword0": 0, "systemfaultword1": 0, "systemfaultword2": 0, "systemfaultword3": 0, "systemfaultword4": 0, "systemfaultword5": 0, "systemfaultword6": 0, "systemfaultword7": 0, "invstartdelaytime": 60, "bdconoffstate": 0, "drycontactstate": 0, "edischrtoday": 0, "edischrtotal": 0, "echrtoday": 0, "echrtotal": 0, "eacchrtoday": 0, "eacchrtotal": 0, "priority": 0, "epsfac": 0, "epsvac1": 0, "epsiac1": 0, "epspac1": 0, "epsvac2": 0, "epsiac2": 0, "epspac2": 0, "epsvac3": 0, "epsiac3": 0, "epspac3": 0, "epspac": 0, "loadpercent": 0, "pf": 10000, "dcv": 0, "bdc1_sysstatemode": 0, "bdc1_faultcode": 0, "bdc1_warncode": 0, "bdc1_vbat": 0, "bdc1_ibat": 0, "bdc1_soc": 0, "bdc1_vbus1": 0, "bdc1_vbus2": 0, "bdc1_ibb": 0, "bdc1_illc": 0, "bdc1_tempa": 0, "bdc1_tempb": 0, "bdc1_pdischr": 0, "bdc1_pchr": 0, "bdc1_edischrtotal": 0, "bdc1_echrtotal": 0, "bdc1_flag": 0, "bdc2_sysstatemode": 0, "bdc2_faultcode": 0, "bdc2_warncode": 0, "bdc2_vbat": 0, "bdc2_ibat": 0, "bdc2_soc": 0, "bdc2_vbus1": 0, "bdc2_vbus2": 0, "bdc2_ibb": 0, "bdc2_illc": 0, "bdc2_tempa": 0, "bdc2_tempb": 0, "bdc2_pdischr": 0, "bdc2_pchr": 0, "bdc2_edischrtotal": 0, "bdc2_echrtotal": 0, "bdc2_flag": 0, "bms_status": 0, "bms_error": 0, "bms_warninfo": 0, "bms_soc": 0, "bms_batteryvolt": 0, "bms_batterycurr": 0, "bms_batterytemp": 0, "bms_maxcurr": 0, "bms_deltavolt": 0, "bms_cyclecnt": 0, "bms_soh": 0, "bms_constantvolt": 0, "bms_bms_info": 0, "bms_packinfo": 0, "bms_usingcap": 0, "bms_fw": 0, "bms_mcuversion": 0, "bms_commtype": 0, "grott_last_push": "2024-08-06T19:23:24.134930+00:00"}
QoS: 0 - Retain: false
Bericht 160 ontvangen op energy/growatt om 21:23:
{"device": "TDG", "time": "2024-08-06T21:23:23", "buffered": "no", "values": {"pvserial": "TDG", "pvstatus": 1, "pvpowerin": 1, "pv1voltage": 867, "pv1current": 0, "pv1watt": 1, "pv2voltage": 1142, "pv2current": 0, "pv2watt": 0, "pv3voltage": 0, "pv3current": 0, "pv3watt": 0, "pv4voltage": 0, "pv4current": 0, "pv4watt": 0, "pvpowerout": 4294967295, "pvfrequentie": 5001, "pvgridvoltage": 2342, "pvgridcurrent": 6, "pvgridpower": 151, "pvgridvoltage2": 0, "pvgridcurrent2": 0, "pvgridpower2": 0, "pvgridvoltage3": 0, "pvgridcurrent3": 0, "pvgridpower3": 0, "vacrs": 2342, "vacst": 0, "vactr": 0, "ptousertotal": 0, "ptogridtotal": 0, "ptoloadtotal": 0, "totworktime": 99151306, "pvenergytoday": 179, "pvenergytotal": 99671, "epvtotal": 103387, "epv1today": 100, "epv1total": 55915, "epv2today": 86, "epv2total": 47472, "epv3today": 0, "epv3total": 0, "etousertoday": 0, "etousertotal": 0, "etogridtoday": 0, "etogridtotal": 0, "eloadtoday": 0, "eloadtotal": 0, "deratingmode": 0, "iso": 65530, "dcir": 0, "dcis": 0, "dcit": 0, "gfci": 0, "pvtemperature": 399, "pvipmtemperature": 399, "temp3": 0, "temp4": 0, "temp5": 486, "pbusvoltage": 3592, "nbusvoltage": 0, "ipf": 20000, "realoppercent": 0, "opfullwatt": 0, "standbyflag": 0, "faultcode": 0, "warningcode": 0, "systemfaultword0": 0, "systemfaultword1": 0, "systemfaultword2": 0, "systemfaultword3": 0, "systemfaultword4": 0, "systemfaultword5": 0, "systemfaultword6": 0, "systemfaultword7": 0, "invstartdelaytime": 60, "bdconoffstate": 0, "drycontactstate": 0, "edischrtoday": 0, "edischrtotal": 0, "echrtoday": 0, "echrtotal": 0, "eacchrtoday": 0, "eacchrtotal": 0, "priority": 0, "epsfac": 0, "epsvac1": 0, "epsiac1": 0, "epspac1": 0, "epsvac2": 0, "epsiac2": 0, "epspac2": 0, "epsvac3": 0, "epsiac3": 0, [... cut ...] _illc": 0, "bdc2_tempa": 0, "bdc2_tempb": 0, "bdc2_pdischr": 0, "bdc2_pchr": 0, "bdc2_edischrtotal": 0, "bdc2_echrtotal": 0, "bdc2_flag": 0, "bms_status": 0, "bms_error": 0, "bms_warninfo": 0, "bms_soc": 0, "bms_batteryvolt": 0, "bms_batterycurr": 0, "bms_batterytemp": 0, "bms_maxcurr": 0, "bms_deltavolt": 0, "bms_cyclecnt": 0, "bms_soh": 0, "bms_constantvolt": 0, "bms_bms_info": 0, "bms_packinfo": 0, "bms_usingcap": 0, "bms_fw": 0, "bms_mcuversion": 0, "bms_commtype": 0}}
QoS: 0 - Retain: false

sirox01, 2024-08-31:
I hope you can help me a little more or tell me witch information I should provide.

typxxi, 2024-09-01:
Back, if you want to get support from anyone consider to put more effort into your requests.

If you do not spend time why should I spend more time ?

on my site the log looks not like this
[image]

In home assistant log of growatt addon I get a well formated output provided by johan.
If you just paste without preparing before to use the <> code option then no one will read nor help. 

For me it loos like a pile of mess. 
Why didn't you use the verbose log option offered in Grott HA addon ?
In such case you get the details needed. 

Formatting your messages and comment is king cause it makes reading easier and therefore people will more likely read and invest time. If you do not care, others do not care. 
And again the level of input from your side is minimal. 
Have you checked if that really happens daily and for just a single sensor only ?

Then it would be very limited and you can spend ages to find the cause or try filter the mqqt.

If you do not care about the time and complexity then no one else will read and invest time, so do not expect much support.

"I have a MIN 3000TL-XE converter"

no,
another example that you do not care enough cause there is no "converter", maybe a hybrind inverter ?
If you do not care to check you message before you send it why should others help you out ?

AGAING: "[THE MARGIN FOR ERRORS IN LIFE IS TOO SMALL](https://youtu.be/1fmZmKsL5eE?si=VW22i5LkNfwhVn7c&t=51)." 
and to understand that watch the video which explains how things work together if you are in 1 team and in a tiny boat to achieve something together -  in this case winning the olympic gold medal in 2008 and what makes the difference between winning and loosing.

Look for the details, show more will to put effort in your topics and comments cause then someone might jump into the boat to row with you to achieve something. 

Copy and paste is not enough,
not checking for errors is another bad habit cause it looks like that others have to do the work for you if you do not care.
If you assume that i must know what was obviously meant then I can say no. I have only my eyes to read, can not see your system or what is wrong and I can only blieve what you are typing and not imagine. I do not even know if that is a hybrid cause the protocol shows hybrid data items but maybe you have no battery right now. 

I am not sure but in case Johan says that everything is mapped (which it should be for the MiN series - I have min 4600, but no min hybrid which is different due to the sensors) then I think he is right, but wrong, that this has something to do with the grott ha addon . Why should such a spike been caused by the grott HA addon ?

And if so wouldn't there be more owners from the min3300 to min6000 all complaining about the same topic since the rollout of the new version ?

What componnent / code is creating and feeding your sensor (usually mqtt messages)?
Are you using Grott HA addon or Muppet3000's Grott integration or did you it your own in yaml?

As i have said before: you have to provide the full picture which i can not. 
And based on the progress I can see so far it might take ages - or someone else who had the same issue will reply.

But I doubt that cause egguy would have heard about that.
And finally: try to setup a grott running on a dedicated pi and observe if there will be the same issues.

I have grott ha addon and muppet3000 integration. It works quite well unless growatt deletes the ip adresses in the stick.

sirox01, 2024-09-02:
I didn't know it exited.

It is the standard Growatt Output sensor from the Grott Addon

I have the log in this message where it is visible it comes from the Grott Addon

What do you mean by converter?

 **Why should such a spike been caused by the grott HA addon ?**
It worked before the summer. After Growatt started to reset ShineLan IP adress the problem occurd.

**And if so wouldn't there be more owners from the min3300 to min6000 all complaining about the same topic since the rollout of the new version ?**
If I knew then there was no problem right?

What componnent / code is creating and feeding your sensor (usually mqtt messages)?
Are you using Grott HA addon or Muppet3000's Grott integration or did you it your own in yaml?
Grott stable branch (2.8)
Current version: 0.1.11
I only have Grott Addon.

```
		 -  loadpercent          :  0.0
		 -  pf                   :  1000.0
		 -  dcv                  :  0
		 -  bdc1_sysstatemode    :  0
		 -  bdc1_faultcode       :  0
		 -  bdc1_warncode        :  0
		 -  bdc1_vbat            :  0.0
		 -  bdc1_ibat            :  0.0
		 -  bdc1_soc             :  0
		 -  bdc1_vbus1           :  0.0
		 -  bdc1_vbus2           :  0.0
		 -  bdc1_ibb             :  0.0
		 -  bdc1_illc            :  0.0
		 -  bdc1_tempa           :  0.0
		 -  bdc1_tempb           :  0.0
		 -  bdc1_pdischr         :  0.0
		 -  bdc1_pchr            :  0.0
		 -  bdc1_edischrtotal    :  0.0
		 -  bdc1_echrtotal       :  0.0
		 -  bdc1_flag            :  0
		 -  bdc2_sysstatemode    :  0
		 -  bdc2_faultcode       :  0
		 -  bdc2_warncode        :  0
		 -  bdc2_vbat            :  0.0
		 -  bdc2_ibat            :  0.0
		 -  bdc2_soc             :  0
		 -  bdc2_vbus1           :  0.0
		 -  bdc2_vbus2           :  0.0
		 -  bdc2_ibb             :  0.0
		 -  bdc2_illc            :  0.0
		 -  bdc2_tempa           :  0.0
		 -  bdc2_tempb           :  0.0
		 -  bdc2_pdischr         :  0.0
		 -  bdc2_pchr            :  0.0
		 -  bdc2_edischrtotal    :  0.0
		 -  bdc2_echrtotal       :  0.0
		 -  bdc2_flag            :  0
		 -  bms_status           :  0
		 -  bms_error            :  0
		 -  bms_warninfo         :  0
		 -  bms_soc              :  0
		 -  bms_batteryvolt      :  0.0
		 -  bms_batterycurr      :  0.0
		 -  bms_batterytemp      :  0.0
		 -  bms_maxcurr          :  0.0
		 -  bms_deltavolt        :  0.0
		 -  bms_cyclecnt         :  0
		 -  bms_soh              :  0
		 -  bms_constantvolt     :  0.0
		 -  bms_bms_info         :  0
		 -  bms_packinfo         :  0
		 -  bms_usingcap         :  0
		 -  bms_fw               :  0
		 -  bms_mcuversion       :  0
		 -  bms_commtype         :  0
	 - MQTT jsonmsg: 
			 {"device": "T", "time": "2024-09-02T20:07:48", "buffered": "no",
			 "values": {"pvserial": "T", "pvstatus": 1, "pvpowerin": 1,
			 "pv1voltage": 785, "pv1current": 0, "pv1watt": 1, "pv2voltage": 813,
			 "pv2current": 0, "pv2watt": 0, "pv3voltage": 0, "pv3current": 0, "pv3watt":
			 0, "pv4voltage": 0, "pv4current": 0, "pv4watt": 0, "pvpowerout": 4294967281,
			 "pvfrequentie": 4999, "pvgridvoltage": 2359, "pvgridcurrent": 6,
			 "pvgridpower": 162, "pvgridvoltage2": 0, "pvgridcurrent2": 0,
			 "pvgridpower2": 0, "pvgridvoltage3": 0, "pvgridcurrent3": 0, "pvgridpower3":
			 0, "vacrs": 2359, "vacst": 0, "vactr": 0, "ptousertotal": 0, "ptogridtotal":
			 0, "ptoloadtotal": 0, "totworktime": 101935705, "pvenergytoday": 126,
			 "pvenergytotal": 103050, "epvtotal": 106906, "epv1today": 71, "epv1total":
			 57808, "epv2today": 61, "epv2total": 49098, "epv3today": 0, "epv3total": 0,
			 "etousertoday": 0, "etousertotal": 0, "etogridtoday": 0, "etogridtotal": 0,
			 "eloadtoday": 0, "eloadtotal": 0, "deratingmode": 0, "iso": 65530, "dcir":
			 0, "dcis": 0, "dcit": 0, "gfci": 0, "pvtemperature": 387,
			 "pvipmtemperature": 387, "temp3": 0, "temp4": 0, "temp5": 475,
			 "pbusvoltage": 3625, "nbusvoltage": 0, "ipf": 20000, "realoppercent": 0,
			 "opfullwatt": 0, "standbyflag": 0, "faultcode": 0, "warningcode": 0,
			 "systemfaultword0": 0, "systemfaultword1": 0, "systemfaultword2": 0,
			 "systemfaultword3": 0, "systemfaultword4": 0, "systemfaultword5": 0,
			 "systemfaultword6": 0, "systemfaultword7": 0, "invstartdelaytime": 60,
			 "bdconoffstate": 0, "drycontactstate": 0, "edischrtoday": 0, "edischrtotal":
			 0, "echrtoday": 0, "echrtotal": 0, "eacchrtoday": 0, "eacchrtotal": 0,
			 "priority": 0, "epsfac": 0, "epsvac1": 0, "epsiac1": 0, "epspac1": 0,
			 "epsvac2": 0, "epsiac2": 0, "epspac2": 0, "epsvac3": 0, "epsiac3": 0,
			 "epspac3": 0, "epspac": 0, "loadpercent": 0, "pf": 10000, "dcv": 0,
			 "bdc1_sysstatemode": 0, "bdc1_faultcode": 0, "bdc1_warncode": 0,
			 "bdc1_vbat": 0, "bdc1_ibat": 0, "bdc1_soc": 0, "bdc1_vbus1": 0,
			 "bdc1_vbus2": 0, "bdc1_ibb": 0, "bdc1_illc": 0, "bdc1_tempa": 0,
			 "bdc1_tempb": 0, "bdc1_pdischr": 0, "bdc1_pchr": 0, "bdc1_edischrtotal": 0,
			 "bdc1_echrtotal": 0, "bdc1_flag": 0, "bdc2_sysstatemode": 0,
			 "bdc2_faultcode": 0, "bdc2_warncode": 0, "bdc2_vbat": 0, "bdc2_ibat": 0,
			 "bdc2_soc": 0, "bdc2_vbus1": 0, "bdc2_vbus2": 0, "bdc2_ibb": 0, "bdc2_illc":
			 0, "bdc2_tempa": 0, "bdc2_tempb": 0, "bdc2_pdischr": 0, "bdc2_pchr": 0,
			 "bdc2_edischrtotal": 0, "bdc2_echrtotal": 0, "bdc2_flag": 0, "bms_status":
			 0, "bms_error": 0, "bms_warninfo": 0, "bms_soc": 0, "bms_batteryvolt": 0,
			 "bms_batterycurr": 0, "bms_batterytemp": 0, "bms_maxcurr": 0,
			 "bms_deltavolt": 0, "bms_cyclecnt": 0, "bms_soh": 0, "bms_constantvolt": 0,
			 "bms_bms_info": 0, "bms_packinfo": 0, "bms_usingcap": 0, "bms_fw": 0,
			 "bms_mcuversion": 0, "bms_commtype": 0}}
	 - Grott MQTT topic used : energy/growatt
	 - MQTT message message sent
	 - Grott send data to PVOutput systemid:  ***** for inverter:  *****
		 -  {'X-Pvoutput-Apikey': '****', 'X-Pvoutput-SystemId': '*****'}
		 -  {'d': '20240902', 't': '20:07', 'v2': 429496728.1, 'v6': 235.9, 'v1': 12600}
```

typxxi, 2024-09-02:
sorry,
I am using the grott integration provided by muppet 3000. 

Mean while i have looked into my sensors and also found 1 that has such big figure, but only sometimes "load" .

You need to catch the message that causes the spike and i guess that is in the log, but I do not know where to find the log.

sirox01, 2024-09-18:
I have build a filter in Homeassistant.
https://www.home-assistant.io/integrations/filter/
Problem is solved

closed 2024-09-18 by sirox01