egguy/addon-grott #75: After update add-on not working anymore

thoompje, 2024-08-01:
Hello All,

After updating this morning the add-on is not working anymore. This is the log:

`Traceback (most recent call last):
  File "/app/grott.py", line 35, in <module>
    proxy.main(conf)
  File "/app/grottproxy.py", line 129, in main
    inputready, outputready, exceptready = ss(self.input_list, [], [])
                                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: filedescriptor out of range in select()
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped`

Anyone any idea?

larrybml, 2024-08-01:
Yes, same problem here. It starts and after short time is stopping itself. Even I start it again , it is stopping itself again, and mqtt does not receive any data. Waiting also for a fix.

egguy, 2024-08-01:
O_o I'm going to take a look. It seems other users are affected: https://github.com/johanmeijer/grott/issues/341

larrybml, 2024-09-04:
Same fault, I just updated again, but same issue , it starts and after that it is shutting down by itself with the folowing log :

ValueError: filedescriptor out of range in select()
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-svscanctl: fatal: unable to control /run/service: supervisor not listening

Just returned back to 0.1.10

closed 2024-10-13 by github-actions[bot]