egguy/addon-grott #81: Where can I find the grott.ini and the log cause I have lost an inverter?

typxxi, 2024-08-31:
Key issue is a missing inverter who seems to send mqtt messages that are not arriving like before.
The usual verbose log is too short cause I have 8 inverters in total so a bit more traffic

In the documentation I have found this line

_**Configuration of grott**
The add-on creates a folder in home assistant. You can find it in the /addon_configs/grott/ folder of your home assistant instance.

You can create/modify the grott.ini using the file editor add-on or VS Code add-on. This allows you to add pvoutput support or to add a custom JSON layout._

I tried to find the grott.ini and found 2 but both of them are empty.
[image]

1st 
the description is a bit misleading cause it should be "creats a folder in your home assistant installation" to avoid peope looking in the "homeassistant" folder for such /addon_configs/grott/ folder.
Not everyone knows that all addon might be stored under root

2nd
the location is not described right or my installation is faulty for whatever reason

I have 
`/addon_configs/1ef..._grott/grott`
but I should have
`/addon_configs/grott/`

And then the grott.ini there is empty.

Of cause I have a 2nd grott.ini for whatever reason in 
`/homeassistant/grott`

which also seems to be empty.

Latest HA from august 2024 (I think 20240803)
I have upgraded to the most recent grott addon 0.1.11

What is wrong ?
1. Where can I find the grott.ini used for my grott addon installation?
2. Where can I find the log file for the mqtt messages and the decoding?

Thanks a lot for the great support.

typxxi, 2024-08-31:
I have checked the growatt datalogger settings over and over again.

This particular inverter showed me the ip adress of my HA like all the other inverters before like 192.168.0.2

BUT: 
I came behind that in case you use the "growatt inverter selector" from the upper left corner to switch to the next inverter to go into  the datalogger settings you have to be aware that you will be fooled !

[image]

If you open the next inverters datalogger settings then you will see the same ip 192.168.0.2 it should have.
But that is not the reality like in my in the beginning 3 missing inverters. I had changed the first one and then checked the others by using the selector and everything looked fine, except mqtt explorer showed just the messages of the first one i had changed. 

Close that chrome register / window with ctrl+w .
Then open the inverter datalogger settings again and then you will see the real ip adress: 

Yes, it is empty. Now type the 192.168.0.2 in and save that. 
Then use the selector again to switch to the next missing one and open the datalogger settings.
You will see the same ip adresse it should have, but if you close the window and open the inverter from the main overview of all the inverters again you will get an empty ip adress. 

This is hard to believe or verify cause I doubt it could be a cache issue. Windows 10 with most recent chrome updated this morning. 

My core issue has been solved

BUT I STILL like to know where my grott.ini is stored or hidden and where I can find the full log that is shown here 
[image]

and btw: there is a wrong release figure "Grott stable branch 2.8" should be "2.83" as I am using HA Grott Addon 0.1.11

typxxi, 2024-10-01:
Any updates where I can find the files ?

egguy, 2024-10-03:
Hi, for the log, it is on the log part of the addon page.

> I have
> /addon_configs/1ef..._grott/grott

As for the `.ini` I confirm it is located in: `/addon_configs/XXXXX_grott/grott/grott.ini` 
[image]

[image]

As for some settings, they can't be set in the `grott.ini`, as the file is read, then the environment settings are applied and override. For those, you need to set them in the add-on config.

[image]

closed 2024-11-11 by github-actions[bot]