egguy/addon-grott #85: Missing Values on MOD 7000TL3-XH with Battery

schlidda83, 2024-09-07:
Hi @egguy 

i´ve already created issues with Johann and Chris. Seems both are unable to help.
here: https://github.com/johanmeijer/grott/issues/587
and
https://github.com/muppet3000/homeassistant-grott/issues/98#issuecomment-2333318576

No Matter what i do enter in the config... i do not get the battrery values unless i enter "MIN" but that is obviously wrong as i have a MOD one.

at the moment i have this set up
I have not entered the Serial of the inverter because i will only receive wrong values then
[image]

therefore the log is saying
`
 - Grott automatic protocol detection
	 - Grott data record length 585
	 - layout   :  T060104X
	 - **no matching record layout found, try generic**
	 - Record layout used :  T06NNNNX
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 04bf000602410104475a4c3044413735453600000000000000000000000000000000000000005
		 1414c30434739304455000000000000000000000000000000000000000018090710180d020bb8
		 0c340001000053e90afc0031000035d20b25001b00001e1700000000000000000000000000000
		 000000054bb000001e000004b95138308fe001c0000192d090b001b0000186a08f0001b000018
		 210f960f880f7400000000000043eb000010d0027a0656000000c400015ab700016b570000008
		 00000c7a5000000700000a3b2000000000000000000000000000000810000008b000011840000
		 004500015a020000000000000000000000f000000000fffa003502d30240000118f9015f02060
		 152000001a00c960c624e20001b00011170000000000000000000000000000000000000000000
		 00003c00000000000100000000000010d0000000d00c350cb10000001a000004e60000003a000
		 00538000000000000000100016f1a0000004600015d9500000000000000000000000000000000
		 00000000000000000000000000000000000000002710000018960000000b0201000002bf3e5c0
		 001006418f60eaf0002000201680000000000be00000000000004e600000538000000020a4700
		 0c00010000013f012c0002000c000000000001000000320032000000000000000000000000000
		 000000000000100040000000000643e58000001c7000009c40000006442903750000000001eeb
		 000000000d020cff3e58000000020000000d0000000002bc5528f8d4000000000000000000000
		 0000000000098fe
	 **- Possible Inverter serial QAL0CG90DU
	 - Inverter serial not recognised - using inverter type default**
	 - Growatt new layout processing
		 - decrypt       :  True
		 - offset        :  6
		 - record layout :  T06NNNNX

	 - Grott data record date/time processing started
	 - date-time:  2024-09-07T16:24:13
	 - Grott values retrieved:
		 -  datalogserial        :  GZL0DA75E6
		 -  pvserial             :  QAL0CG90DU
		 -  pvstatus             :  1
		 -  pvpowerin            :  2148.1
		 -  pv1voltage           :  281.2
		 -  pv1current           :  4.9
		 -  pv1watt              :  1377.8
		 -  pv2voltage           :  285.3
		 -  pv2current           :  2.7
		 -  pv2watt              :  770.3
		 -  pvpowerout           :  1934.9
		 -  pvfrequentie         :  50.0
		 -  pvgridvoltage        :  230.2
		 -  pvgridcurrent        :  2.8
		 -  pvgridpower          :  644.5
		 -  pvgridvoltage2       :  231.5
		 -  pvgridcurrent2       :  2.7
		 -  pvgridpower2         :  625.0
		 -  pvgridvoltage3       :  228.8
		 -  pvgridcurrent3       :  2.7
		 -  pvgridpower3         :  617.7
		 -  totworktime          :  5771.0
		 -  pvenergytoday        :  19.6
		 -  pvenergytotal        :  8875.9
		 -  epvtotal             :  9301.5
		 -  epv1today            :  12.8
		 -  epv1total            :  5110.9
		 -  epv2today            :  11.2
		 -  epv2total            :  4190.6
		 -  pvtemperature        :  35.1
		 -  pvipmtemperature     :  41.6
	 - MQTT jsonmsg: 
			 {"device": "QAL0CG90DU", "time": "2024-09-07T16:24:13", "buffered": "no",
			 "values": {"datalogserial": "GZL0DA75E6", "pvserial": "QAL0CG90DU",
			 "pvstatus": 1, "pvpowerin": 21481, "pv1voltage": 2812, "pv1current": 49,
			 "pv1watt": 13778, "pv2voltage": 2853, "pv2current": 27, "pv2watt": 7703,
			 "pvpowerout": 19349, "pvfrequentie": 4995, "pvgridvoltage": 2302,
			 "pvgridcurrent": 28, "pvgridpower": 6445, "pvgridvoltage2": 2315,
			 "pvgridcurrent2": 27, "pvgridpower2": 6250, "pvgridvoltage3": 2288,
			 "pvgridcurrent3": 27, "pvgridpower3": 6177, "totworktime": 41551446,
			 "pvenergytoday": 196, "pvenergytotal": 88759, "epvtotal": 93015,
			 "epv1today": 128, "epv1total": 51109, "epv2today": 112, "epv2total": 41906,
			 "pvtemperature": 351, "pvipmtemperature": 416}}
	 - Grott MQTT topic used : energy/growatt
	 - MQTT message message sent
	 - Grott Send data to PVOutput disabled 
	 - Grott Send data to Influx disabled 
	 - Grott extension processing disabled 
	 
`

System
HomeAssistant on a Synology as docker
grott via HACS
grott custom components from muppet (https://github.com/muppet3000/homeassistant-grott)

do you need any more info?

typxxi, 2024-09-08:
even though I think that I had read somewhere that the parameters for the inverter mapping should be uppercase I suggest to give "tl-3" a try and then to setup a real grott installation in parallel on a pi 3 or what soever.

You then have to reconfigure the datalogger stick to the ip of the pi 3 with grott (Pi OS)
And then you should be able to get the sensor data from that machine and we would need to compare the logs if the behaviour is still the same or not.

I am expecting the same behaviour cause afaik egguy is not dealing with the decoding template or however they are called. He is simply converting GROTT into an HA Addon and he is not developing his own templates. 

If you can show that the grott installation would behave the same then it would be Johans job to look a bit closer. RIght now the ball has been played to egguy but I guess you would get a fast stolution with a basic GROTT setup cause then Johan would support you. VM or another PC or a small pi is enough to run grott, even Pi 2 should be capable if I remember right my beginnings with grott a year ago.

typxxi, 2024-09-09:
I have called a friend of mine who has had such issues last year before the next grott stable release had been released and when a lot of templates were missing like min 4600 or so. 

He said that he got that running with a grott setup from scratch on an old Pi 2 or so.
Back then he had been guided by Johan Meier and everything was fine within hours or so. 
Many months later and when I had already migrated to HA he finally migrated to egguys AddOn too in may or june.

egguy, 2024-09-10:
I will generate a new build with the new JSON layout. I will keep you updated.

Sorry for the delayed response I'm in vacation

typxxi, 2024-09-11:
Thanks and enjoy the holidays - as we can offer rain and sunshine, wind, clouds and temps from 15 to 30° changing from day to day.

typxxi, 2024-09-15:
Could you let us more about what went wrong or why it is different (or the same) compared to grott standalone ?

What does new json layout mean, a developement by you or simply upgrading what JOhan had called a template or so ?

thanks

egguy, 2024-09-16:
The difference is mostly a size optimisation, there is no difference between the code of grott standalone and the addon version.

I was only copying the original source code without the extra in the container to make it smaller (the .py files).

Grott has the ability to load extra inverter layout (mapping of content of packet to values) from JSON files, but I wasn't including it as my assumption all the layouts were already included in the python code.

Some inverters now, like the MOD, are supported by layout defined in the JSON files, e.g https://github.com/johanmeijer/grott/commit/3fa7ef970a48d29d477898d4a6041c9e948eda40

I just need to include the file during the build process

typxxi, 2024-09-16:
> Grott has the ability to load extra inverter layout (mapping of content of packet to values) from JSON files, but I wasn't including it as my assumption all the layouts were already included in the python code.
> 
> Some inverters now, like the MOD, are supported by layout defined in the JSON files, e.g [johanmeijer/grott@3fa7ef9](https://github.com/johanmeijer/grott/commit/3fa7ef970a48d29d477898d4a6041c9e948eda40)

ah, and that was a very long ahhhhhh
that I can understand the mystery which also could have been found out by setting up - as I had recommanded - a dedicated pi grott server to solve the question if that is caused by grott or grott ha addon. 

Thanks a lot for solving and explaining the details. 
I have not yet read johan meiers layout defined json files topic but I guess it becomes more important and could be good to know as I have so many and also different models, just got another SPH4000 and SPH 4600, both used for a few months.

Thanks

typxxi, 2024-09-17:
> Some inverters now, like the MOD, are supported by layout defined in the JSON files, e.g [johanmeijer/grott@3fa7ef9](https://github.com/johanmeijer/grott/commit/3fa7ef970a48d29d477898d4a6041c9e948eda40)
> 
> I just need to include the file during the build process

OK, 
I have looked for the file , which was in the root directory for whatever reason, while a lot of others can be found [here](examples/Record Layout) under examples and then Record Layout 

I assume that these "inverter model layout" config files listed here are required by grott and that the missing MOD file or maybe the latest version of the MOD file was missing.  

I do  not have that mod or TL3 inverter, but for sure good to know how things work together cause I had seen different of these layouts in the log which can look very different in the look based on the functioniality of the inverter. 

Let me know if I am wrong - and yes , I have not forgotten to help you in 2025 with the documentation parts. That's why I am asking cause still grott can be a big guess work at least for me for example when I had lost one inverter after the other in the past 3 weeks simply due to maintenance during daylight which then "calls the Growatt Cloud back into action" and overwrote my IP adresses again, but only 1 inverter at a time, not all or just some. So far I had only seen that I lost 1 after reboot and usually a different one than before, therefore my assumptions that the growatt cloud empire strikes back. 
Screenshot of the other layour records. 

[image]

Final question: where is grott addon HA stored on a HA OS system ?
I can not find the grott.ini file or the one that is used cause those I had found were grott.ini with a size of 0
They were empty. But I did not find anything , no grott directory with files.

egguy, 2024-09-30:
Hi @schlidda83 I have finally released the add-on with layout on the edge branch: https://github.com/egguy/ha-addons-edge/

You can add the edge repository, stop the stable release, copy your config over to test, and start the edge version.

In case of success, you should see in the log the mention of :
- T065020
- T06NNNNXMOD

If you could confirm me everything is working as intented. I will promote this to the stable release

egguy, 2024-09-30:
> Final question: where is grott addon HA stored on a HA OS system ?

If i'm not mistaken, the add-on run in its container. You don't have direct access. I think you can have a better overview by using the portainer add-on

The docker container then mount a custom folder : https://developers.home-assistant.io/docs/add-ons/configuration#add-on-configuration to store the addon configuration. This is where I store the grott.ini (empty by default) you can  modify.

> I assume that these "inverter model layout" config files listed here are required by grott and that the missing MOD file or maybe the latest version of the MOD file was missing.

Yes and not. Some layouts are defined in the code (like the TL3, see grottconf.py: https://github.com/johanmeijer/grott/blob/Master/grottconf.py#L1007) so they are already included. Other like the `MOD` are not defined in the code, they are loaded from the JSON files at the start if they are present in the current directy where grott run.

egguy, 2024-10-03:
The config should be good enough.

One weird thing is the packet you show as an example is shorter than expected, this makes grott use the generic layout instead of the specific one. The returned value seems to make sense.

schlidda83, 2024-11-03:
Hi @egguy 
i have seen that the standard growatt  server addon is receiving a lot of more information. how can that be?

egguy, 2024-11-08:
Hi @schlidda83, what is the do you mean by the growatt server add-on?

I need to take a look if the MOD series is not part of the new beta branch of grott. This could explain why it is only giving a limited set of informations.

FlexxFR, 2024-12-08:
Hi @egguy , I'm hitting the same issue here with a MOD 8000TL3-XH invertor. All strange values.
I all have latest (regular) versions installed, tried to use the parameter tl3, TL3, min, MIN, mod, MOD, but nothing works.
(Or better, diagnostic values are okay, sensor values give weird numbers)

Reading the comments above, I'm not sure what to do. Do I need to install another version (-edge?)
And then use the MOD as type?

closed 2025-01-16 by github-actions[bot]