egguy/addon-grott #88: Support for Growatt ShineWifi X2

jafar-atili, 2024-09-12:
I have just installed a Growatt Hybrid inverter.

I received the ShineWifi X2 and the server configuration doesn't seems to allow changing the port from 7006:

[image]

I tried changing the Grott Addons port to 7006 but it still comes up either tje default port (5279)

And I am not sure if the integration even expecting an MQTT packet...

egguy, 2024-09-13:
Hi,

The addon system in HA allows you to do port redirect without reconfiguration of grott. You can change the port in the port configuration: 
[image]

And even if grott still displays 5279, the HA interface will listen on 7006 and forward to the 5279 port in the addon.

As for this stick and MQTT I don't think this add-on process MQTT.

jafar-atili, 2024-09-14:
I did that and I got invalid packet in the log...

closed 2024-10-23 by github-actions[bot]