egguy/addon-grott #103: Need help with latest update, it is stopping

larrybml, 2024-10-23:
Hello, my fact is this : 
Home assistant on pi4
Grott stable branch (2.8) 0.1.10
Mosquitto broker 6.4.1
Grott 1.1.0
Inverter : Growatt MOD 10KTL3-X

Everything is working OK with this setup. BUT, if I want to update Grott stable branch (2.8) to 0.1.11 or to newer 0.1.12 , Grott stable branch (2.8) stops working. Is trying to start 1-2 times, and after that it is stopping itself and my setup is not working anymore.
Can someone please help me with this issue ? 
Thank you in advance.

typxxi, 2024-10-23:
I have upgraded this morning to 2.8 and version 0.1.12 as you can see.
[image]

It is running smoothless so far and all of my 8 inverters are working fine, of which you can see 6 here in my screenshot
[image]

But if I remember correctly the last upgrade were targetting the MOD series which seem to have some issues with the proper layout. 

Do you have watchdog enabled or not ?

My growatt addon running on 0.1.12 has not rebooted but I have only a bunch of SPH , MIN and MiC inverters in use, but not a single growatt mod model.

I guess you have to wait for the other mod owners that they have further hints.

thoompje, 2024-10-23:
Same here, I got the following error.

```

	 - Growatt packet received:
		  <socket.socket fd=324, family=2, type=1, proto=0, laddr=('172.30.33.8', 33204), raddr=('172.25.1.7', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=326, family=2, type=1, proto=0, laddr=('172.30.33.8', 33210), raddr=('172.25.1.7', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=328, family=2, type=1, proto=0, laddr=('172.30.33.8', 33218), raddr=('172.25.1.7', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=330, family=2, type=1, proto=0, laddr=('172.30.33.8', 33226), raddr=('172.25.1.7', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=332, family=2, type=1, proto=0, laddr=('172.30.33.8', 33228), raddr=('172.25.1.7', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=334, family=2, type=1, proto=0, laddr=('172.30.33.8', 33232), raddr=('172.25.1.7', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=336, family=2, type=1, proto=0, laddr=('172.30.33.8', 33234), raddr=('172.25.1.7', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=338, family=2, type=1, proto=0, laddr=('172.30.33.8', 33240), raddr=('172.25.1.7', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=340, family=2, type=1, proto=0, laddr=('172.30.33.8', 33256), raddr=('172.25.1.7', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=342, family=2, type=1, proto=0, laddr=('172.30.33.8', 33264), raddr=('172.25.1.7', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=344, family=2, type=1, proto=0, laddr=('172.30.33.8', 33276), raddr=('172.25.1.7', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=346, family=2, type=1, proto=0, laddr=('172.30.33.8', 33290), raddr=('172.25.1.7', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=348, family=2, type=1, proto=0, laddr=('172.30.33.8', 33306), raddr=('172.25.1.7', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=350, family=2, type=1, proto=0, laddr=('172.30.33.8', 33312), raddr=('172.25.1.7', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=352, family=2, type=1, proto=0, laddr=('172.30.33.8', 33322), raddr=('172.25.1.7', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=354, family=2, type=1, proto=0, laddr=('172.30.33.8', 33332), raddr=('172.25.1.7', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=356, family=2, type=1, proto=0, laddr=('172.30.33.8', 33336), raddr=('172.25.1.7', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=358, family=2, type=1, proto=0, laddr=('172.30.33.8', 33338), raddr=('172.25.1.7', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=360, family=2, type=1, proto=0, laddr=('172.30.33.8', 33354), raddr=('172.25.1.7', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=362, family=2, type=1, proto=0, laddr=('172.30.33.8', 33364), raddr=('172.25.1.7', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=364, family=2, type=1, proto=0, laddr=('172.30.33.8', 33380), raddr=('172.25.1.7', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=366, family=2, type=1, proto=0, laddr=('172.30.33.8', 33390), raddr=('172.25.1.7', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=368, family=2, type=1, proto=0, laddr=('172.30.33.8', 33402), raddr=('172.25.1.7', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=370, family=2, type=1, proto=0, laddr=('172.30.33.8', 33408), raddr=('172.25.1.7', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=372, family=2, type=1, proto=0, laddr=('172.30.33.8', 33422), raddr=('172.25.1.7', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=374, family=2, type=1, proto=0, laddr=('172.30.33.8', 33430), raddr=('172.25.1.7', 5279)>

	 - Growatt packet received:
		  <socket.socket fd=376, family=2, type=1, proto=0, laddr=('172.30.33.8', 33438), raddr=('172.25.1.7', 5279)>
Traceback (most recent call last):
  File "/app/grott.py", line 35, in <module>
    proxy.main(conf)
  File "/app/grottproxy.py", line 129, in main
    inputready, outputready, exceptready = ss(self.input_list, [], [])
                                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: filedescriptor out of range in select()
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

```

Anyone a idea?

larrybml, 2024-10-24:
Exactly like you, I have the same problem with 0.1.11 and 0.1.12 updates. It is trying to start, it is complaining about some less data received that is expecting , about some lybscr not installed and after that it is stopping successfully like yours.
But why with 0.1.10 it is working fine?
Do we need to set something else more?

@typxxi it is very nice that for you is working, the thing is that for some is working for other no, and ... this is a very strange thing. As long as it is working perfect with old version, what is changed, or what needs more the new versions that are not working.

If someone knows how to solve this problem, please help, I can provide remote , so maybe someone can figure it out.

thoompje, 2024-11-23:
This is still a issue....

closed 2025-01-08 by github-actions[bot]