egguy/addon-grott #142: Got unexpected response from the API: Service not enabled

TheRealWiesel87, 2025-04-11:
Since 11.04.2025 the Add-ON stops shortly after starting with the following error message:

Got unexpected response from the API: Service not enabled

here is the excerpt from the log:

Add-on: Grott stable branch (2.8)
 The Growatt inverter monitor with current HA plugin integrated
-----------------------------------------------------------
 Add-on version: 0.1.12
 You are running the latest version of this add-on.
 System: Home Assistant OS 15.1  (amd64 / qemux86-64)
 Home Assistant Core: 2025.4.1
 Home Assistant Supervisor: 2025.04.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[07:54:26] INFO: Preparing to start...
[07:54:27] ERROR: Got unexpected response from the API: Service not enabled
./script.sh: line 239: GROTT_CONFIG_MQTT_HOST: unbound variable
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting

rcvaneck, 2025-05-07:
After rebooting the Mosquitto broker add-on in HA is it working for me.

TheRealWiesel87, 2025-05-07:
This also helped me once, but after a few days it stopped working again. Sometimes it worked again on its own, but only for a few days. I have now configured it so that no more data is sent to Growatt in the cloud, and since then it has been working perfectly. Here is my config as yaml: 

`data_path: /config/grott
gmode: proxy
grott_mqtt: true
ginvtype: sph
ginvtypemap: ""
retain: false
verbose: false
ha_plugin: true
mqtt: {}
noipf: true
senddata: false
blockcmd: true
mqttbroker: IP-Address
mqttport: 1883
mqtttopic: growatt
mqttuser: username
mqttpassword: passwort`